RedBearLab/CCLoader #1: cc loader is not opening

shehroooo, 2016-05-10:
when i clicked to ccloader.exe  ,some kind of window appears for less than a second and then suddenly disappears. Somehow i managed to take screenshot of that...pls tell me what is this error says.

thank you in advance:)
[image]

XuGuohui, 2016-05-11:
The CCLoader.exe should be executed from command line terminal, rather than by clicking it to run.

Cheers!

shehroooo, 2016-05-11:
How ? Can u tell me pls .. As i m new to this . pls

XuGuohui, 2016-05-12:
According to the screenshot, you should start the command line prompt first and then change the work path to where CCLoader located:
`cd Desktop\CCLoader-master\Windows`
Then type `CCLoader.exe` and press Enter to execute this program. It will print some message to guide you how to pass parameters to it.

Cheers!

shehroooo, 2016-05-12:
Where do i place .bin file ? N how i locate the .bin file in comand prompt.

All things are going well. Thanks to your guidence ,But now the error says file is not existed or is occupied

shehroooo, 2016-05-12:
[image]

shehroooo, 2016-05-12:
[image]

closed 2016-05-12 by shehroooo

XuGuohui, 2016-05-13:
The bin file is the one you are going to load to BLEMini and you should type this file's name with its path, either relative or absolute.

shehroooo, 2016-05-13:
How ? Can u please show ... Thanks in advance . :)

XuGuohui, 2016-05-13:
E.g. `CCLoader.exe 9 ..\Bin\Demo.bin 0`.

shehroooo, 2016-05-13:
[image]

shehroooo, 2016-05-13:
[image]

shehroooo, 2016-05-13:
Thank you so much uploader for the help.....cheers :)