RedBearLab/CCLoader #6: Error message : waiting for response from arduino.

palaniappan10, 2016-10-30:
I have connected arduino 5,6,4 and BLE 7,8,11 pin respectively and I have flashed the ccloader sketch to my arduino. But when I load the bin file ccloader.exe stuck at waiting for response from arduino.. Any help

XuGuohui, 2016-10-31:
Are you using Leonardo or else?

palaniappan10, 2016-11-02:
No I used arduino pro mini and nano both result in the same

XuGuohui, 2016-11-02:
Make sure that the serial object communicating with PC is `Serial` on the Arduino side, since we use `Serial` to receive data from PC. If not, you have to adapt the serial object in the sketch.

Also, make sure the connection between Arduino and BLE Mini is well connected as the [README](https://github.com/RedBearLab/CCLoader).

iw2lsi, 2018-11-15:
Same problem here... 

   the application stuck waiting for a reply:

```
~\Windows>CCLoader.exe 6  CC2541hm10v540.bin 0
Copyright (c) 2013 RedBearLab.com
CCLoader.exe version 0.5
Comport : COM6
Bin file: CC2541hm10v540.bin
Device  : Default (e.g. UNO)

Comport open!
<Baud:115200> <data:8> <parity:none> <stopbit:1> <DTR:off> <RTS:off>

File open!
Block total: 512

Enable transmission...
Request sent already!
/********************************************************************/
* If there is no respond last for 3s, please press "Ctrl+C" to exit!
* And pay attention to :
*   1. The connection between computer and Arduino;
*   2. The connection between Arduino and CC2540;
*   3. Whether the device you using is Leonardo or not;
*   4. Other unexpected errors.
/********************************************************************/

Waiting for respond from Arduino...
```
I'm using an Arduino Pro Mini at 3.3V with no voltage converter... all the connections have been checked and I've try with different usb/serial adapter... 

        any ideas ?

             GIampaolo

iw2lsi, 2018-11-15:
mmm... seems that my Arduino Pro Mini is at 8MHz instead of 16...

Jeansidharta, 2021-03-25:
For anyone having this problem, I was using an Arduino Uno with CC2541 and a Raspberry PI 4b. I was having the same problem (stuck at `Waiting for respond from Arduino...`).

My solution was to comment [this](https://github.com/RedBearLab/CCLoader/blob/master/Arduino/CCLoader/CCLoader.ino#L529) line and run CCLoader with 1 as the last argument

ViorelX, 2022-05-31:
Hello!
I have the same problem. It doesn't help to comment that line and the last argument should remain 0 for me, since I have a regular Arduino Uno.