RedBearLab/CCLoader #7: CCloader - how to use it ?

Netoperz, 2016-12-07:
Hello, stupid question, but i can't find any info about construction of command line command to start flashing.

I have got Leonardo with sketch uploaded, and CCloader.exe and HCI_UART_115200bps_20130429.bin

And i want to flash it. My Leonardo is on on COM12

Sorry for that question, but no idea where to put switch for com port and i do not want to damage the CC2540 IC by using flasher in a wrong way.

XuGuohui, 2016-12-08:
You can simply type CCLoader.exe and press Enter, it will print prompt information about the usage.

Netoperz, 2016-12-08:
Thank You, in deed  it works, there is a bug in win 10, when you just drop the .exe to the terminal window sometimes it adds the path to file twice, i didn't noticed that, and that was the main problem.

Usage: C:\Users\miros\TI\CCLoader.exe <com number> <bin file> <device>
Example: C:\Users\miros\TI\CCLoader.exe 2 abc.bin 0
 <device>: 0 -- Default (e.g. UNO)
           1 -- Leonardo

so everything is fine.