RedBearLab/CCLoader #9: Windows 10 64-bit - CCLoader.exe not compatible

keith721, 2017-03-23:
Thank you for posting the C source code for the Windows/Dos executable program.  I have built it using the [TDM-GCC](https://sourceforge.net/projects/tdm-gcc) compiler and successfully flashed my CC41 using it.  If you'd like to host it here, I'm attaching 
[CCLoader_x86_64.zip](https://github.com/RedBearLab/CCLoader/files/866428/CCLoader_x86_64.zip)

the executable for others to download.

XuGuohui, 2017-03-24:
Hi @keith721 , does the program have any dependencies when running ?

keith721, 2017-03-24:
No dependencies of which I'm aware.  I'll uninstall the TDM-GCC package and re-run to verify.

XuGuohui, 2017-03-27:
Awesome！ Thank you for your contribution!

keith721, 2017-03-27:
I'm not having any luck creating a new pull request.  For your information, the commit of this x86_64 executable to my fork is 321b82d82b65ffc1928c5518acd914ae50de113d

Chris01cz, 2017-03-27:
Original ccloader.exe works for me on W10 Pro x86_64.

closed 2017-03-28 by keith721