RedBearLab/CCLoader #13: CCLoader is stuck at the first block

Zap123, 2017-07-22:
Hi, I'm trying to flash 2 different BLE module(JDY-08 and MLT-BT05) as described in the arduino [thread](http://forum.arduino.cc/index.php?topic=393655.0) but CCLoader get stuck at the first block. I have tried both with Linux(never get past "waiting for arduino") and windows.
Both modules have become unusable after the procedure (they are in debug state(?)) so it must have done something.
I'm using an Arduino Pro Mini at 3.3v connected with an USB TTL converter to the PC.
Is there something I can do? Thanks

Zap123, 2017-07-25:
I solved it changing the baudrate both in the sketch and in the windows source to 57200. The original speed was too fast for my arduino.

closed 2017-07-25 by Zap123

altgugos, 2017-12-02:
How did you change the baud? I could'nt do it no matter what I do

Zap123, 2017-12-02:
https://github.com/RedBearLab/CCLoader/blob/master/SourceCode/Windows/main.c#L102
Here, I put 57200 just to be safe and recompiled the source.

altgugos, 2017-12-02:
It should be 57600. 
Anyway, I compiled it again but no solution. Program waits a respond from Arduino. I use Pro Mini and set 0 for the device parameter. Is it correct?

Zap123, 2017-12-02:
Yes 57600 sorry, and 0 is correct. 
It has been a while since I tried, so don't remember much.
Have you put 57600 also here in the sketch?
https://github.com/RedBearLab/CCLoader/blob/master/Arduino/CCLoader/CCLoader.ino#L526