RedBearLab/CCLoader #14: Is it possible to use an Arduino DUE or Zero?

carlosdelfino, 2017-09-01:
From the little that I have analyzed of the code, there is no exclusive nail of the Arduino UNO, therefore it seems perfectly possible to use an Arduino DUE, but before starting, I would like to have this answer on your part. What do you think? do you see something that implies that such a code is used directly on Arduino DUE or Zero? And even if you think it's complex to carry it?

The advantage of using DUE or Zero is the voltage compatibility.

FunnyKiwi, 2017-09-23:
The exe file will instantly close if you try to upload to a DUE, I tried it and it said "invalid parameter". It seems as if it only supports the UNO and Lenardo.

linpp, 2017-12-24:
With some minor changes to the sketch, I got it to run on a ESP8266, which is not even a real Arduino, so I would think you could get it to run on a DUE or Zero.  You just need to map the DD, DC, RESET and LED pins to correct pins on your board, and make sure you are using the right serial port (for dual USB boards).  The only reason the windows program would return "invalid parameter" is if you didn't specify all the arguments.  The UNO/Leonardo switch is not optional, it really just means whether to assert DTR (0 | 1) on the serial port based on the board you are using.

nullp0tr, 2018-01-25:
I haven't tried with a DUE or a ZERO, but I used a ESP8266 as well, and even a NANO, so it's definitely doable without much fuss, Just change the pinout as needed and check the DTR on the board.

DirtyEngineer, 2018-01-30:
@nullp0tr @linpp 
May I inquire as to what pins/changes you use to get the ESP8266 to work?

I wired up a Wemos D1 Mini using pins
 DD = 6;
 DC = 5;
 RESET = 4;
 LED = LED_BUILTIN;

CCloader doesn't get a response from the Wemos.  Using the correct com port.  I'm sure it's something minor I missed when compiling the code for the Wemos. Any help would be greatly appreciated.

linpp, 2018-01-30:
@DirtyEngineer 
I'm not sure what pins you are hooking up to, but using raw numbers for pins means you have to do the board mapping in your head.  You are using GPIO 4, 5, 6 and GPIO 6 isn't even available on any devkit. For ESP, you need to be using the D# pin mappings defined in pins_arduino.h for your board.

Note that on an ESP devkit, D4 is the same as LED_BUILTIN (GPIO 2), so you cannot use D4 for this sketch.  Here's the mapping I used for a v3 devkit, it should also work for your WeMos D1 mini:

int DD = D3;
int DC = D2;
int RESET = D1;
int LED = LED_BUILTIN;

DirtyEngineer, 2018-01-30:
@linpp 
Well that was a silly mistake for me to make wasn't it (facepalm). I should have caught that.

Thank you for bearing with me and taking the time to clarify this. Much appreciated. 

Cheers