RedBearLab/CCLoader #15: DD and DC

KennetZed, 2018-01-22:
I've been googling for ages! can't seem to find pin-info about this hm-11 board from ebay.

Any idea which is DD and DC on this?
(this one is broken - please disregard that)

[image]

nullp0tr, 2018-01-25:
A view of the board from the other side might help.

KennetZed, 2018-01-26:
You're right. A picture of the front:
[image]

nullp0tr, 2018-01-26:
Hmm, that doesn't look like a CC254x. A google search is telling me this is a TLSR8266, not sure how compatible it is with a CC254x.

KennetZed, 2018-01-26:
Darn it! Seems like both a british seller and a chinese seller have fooled my friend who bought them for this project :-) 
He even purchased from two different ebay-sellers, and it seems ended up with the wrong ones twice - i now see that other sellers call these exact ones "compatible CC2541", and not the real thing.

Thank you for your help! we'll better try to purchase new ones from different sellers, and look closely at the images, to make sure we get the correct ones.

nullp0tr, 2018-01-26:
I guess you can close the issue then:) good luck on your project!

closed 2018-01-26 by KennetZed