RedBearLab/CCLoader #16: ESP8266 Support

denxhun, 2018-02-15:
Is this project supports ESP8266 boards - instead of Arduino? I ask because the ESP is a 3.3V based board and hence no level converter is needed.

I tried your great tool with a WeMos D1 mini: compiled, uploaded but "Waiting for respond from Arduino..."
(Tried with UNO and Leonardo options too.)

jeanniquini, 2018-11-12:
Works with ESP8266! I used a NodeMCU v2.

ptath, 2019-03-22:
> Works with ESP8266! I used a NodeMCU v2.

Great, which pins of NodeMCU was used?

jeanniquini, 2019-03-22:
> > Works with ESP8266! I used a NodeMCU v2.
> 
> Great, which pins of NodeMCU was used?

https://github.com/jeanniquini/CCLoader

ptath, 2019-03-22:
> > > Works with ESP8266! I used a NodeMCU v2.
> > 
> > 
> > Great, which pins of NodeMCU was used?
> 
> https://github.com/jeanniquini/CCLoader

Thanks, but no luck =(

```
MBP:CCLoader-master ptath$ ./a.out /dev/cu.wchusbserial640 CC2531ZNP-Prod.bin 
Opening comport...
Comport open!
Read portsettings...
Changing portsettings...
Baud:115200 data:8 parity:none stopbit:1 DTR:off RTS:off
File open success!
Block total: 512
Waiting for Arduino setup...
Remain: 5
Remain: 4
Remain: 3
Remain: 2
Remain: 1
Enable transmission...
Request sent already!Waiting for respond...
Begin programming...
^C
```

jeanniquini, 2019-03-23:
@ptath 
Did you follow all the steps?
You need the Arduino IDE.

ptath, 2019-03-23:
All steps, sure. Flashing by atom+platformio, it is ok. 

If I try to change pins (both sketch and on board), sometimes flashing starts (1-7 or 1-9) and stops with verification error.

jeanniquini, 2019-03-24:
I do not know what it can be, it worked perfectly here. I used NodeMCU v2.
Check the weld on the BLE module.
Use CCLoader_x86_64.exe

h1tch007, 2020-08-24:
> Thanks, but no luck =(
>
> ```
> MBP:CCLoader-master ptath$ ./a.out /dev/cu.wchusbserial640 CC2531ZNP-Prod.bin
> Opening comport...
> Comport open!
> Read portsettings...
> Changing portsettings...
> Baud:115200 [...]

Had the same issue, changing NodeMCU 0.9 to 1.0 in Arduino IDE and then flashing CCLoader resolved problem.
But there was errors in CCLoader when I upload firmware via macOS/linux, successfully uploaded only with Windows binary