RedBearLab/CCLoader #25: Try to execute CCLoader.exe. Nothing happens

ygreq, 2020-03-04:
Hi there! I am trying to execute CCLoader.exe. I double click on it. Run option appears. I run it. Nothing happens. I also tried CCLoader_x86_64.exe and also tried to run both via Run as administrator. Still nothing. 

Any idea what I should do?
Thank you!!

funky2x, 2020-03-07:
HI ygreq.
Move the CCloader to  C directory for easy access from your command prompt window.
On your keyboard hit Windows logo and R and in the Box type cmd.
Now in the command prompt windows type cd.. and hit enter, repeat it.
Now you should see , C:\> type here ...... CCloader.exe and hit enter . now you will see a response to your request.

NorwinA, 2020-04-15:
C:\Users>CCloader.exe
'CCloader.exe' is not recognized as an internal or external command,
operable program or batch file.

stevefoxuk, 2020-04-16:
> C:\Users>CCloader.exe
> 'CCloader.exe' is not recognized as an internal or external command,
> operable program or batch file.

If you have put the CCLoader.exe in the route of C drive then your command is wrong and that is why its failing.

if your CCLoader file is in the very first place you see when opening C drive then you need to change the directory that CMD is in before executing the command.

your command line is saying C:\users and not C:\  

C:\ is the route directory of C drive.

the command to change to the route of C drive is     CD \

im not an expert on command line and i cant help you any further than that but that appears to be the reason why your command is not being recognised as your giving the command for it to look in  C:\users  and not C:\

wangpengabc, 2020-07-06:
[image]
You should input some parameters following the exe file. You can have a look at the picture that I show and I think that you will get the answer. Hope help for you.