RedBearLab/CCLoader #35: CC2531 flashed with Wemos D1 Mini successfully - but dongle not working

Jannomag, 2021-07-25:
Hi,
I tried to flash my new CC2531 dongle via a Wemos D1 Mini with [this ](https://www.zigbee2mqtt.io/information/alternative_flashing_methods.html)tutorial, I followed.

The wiring is correct and the flash process went through successfully without any issues.
But after flashing the dongle stopped working. The onboard LED is off and nothing happens when I stick it into my computer. It also won't show up in Windows' device manager or in 'lsusb' under Linux.

So I tried flashing the stick again and it flashed successfully as well...but still, the dongle doesn't work.

Any ideas how to fix this?

Avamander, 2021-08-22:
For me the process worked and I used an ESP32 with CCLoader. The dongle enumerates.

Try some other binary?

Avamander, 2021-08-22:
Yeah, other binary worked. Check out the linked issue for a potential solution.