RedBearLab/CCLoader #37: Can CCLoader  run on esp32?

Tolomaj, 2023-06-08:
Is CCloader runing properly on esp32?

Sketch uploads to esp32 jist fine.

But can here be any problem with this setup ?

Tolomaj, 2023-06-08:
For others i tried and yes and it works on esp32

EntryGoon, 2023-10-10:
I'm trying using ESP32 instead of ESP8266. can ypu please let me know on which pins of the ESP32 I should connect ? it should be GPIO4, GPIO5 and GPIO12 for ESP8266, but I assume GPIO are not the same for ESP32...

Tolomaj, 2023-10-13:
Sorry.
It was few months since I played with Ccloader, and I don't remember anymore witch pins that was. 
But after quick look in fork of original repo for esp8266([link to repo](https://github.com/jeanniquini/CCLoader/tree/master)) i find this.
[image]
i assume you can set your pins in .ino file.

EntryGoon, 2023-10-16:
Yep i saw that and I have to try this. Thanks a lot for your help :-)

ds2k5, 2023-11-20:
@EntryGoon 

Hi could you be so kind and help me out....
Did not get FW flashed to CC2530...

Tolomaj, 2023-11-23:
As i write above, I dont remember exactly how flash the chip. But..

In my case i flash cc254x chip (actually HM-10 but chip should be same).
Because the datasheet is same for CC254x and CC2530 i think flashing should work the same.

I asked Google and found some thinks i remeber on site: https://www.hangar42.nl/ccloader.
Like pins used for flashing. (I know this is arduino but on CC side it doesn't matter).
[image]
And I assume the esp32 pins is this:
> [image] 

Also, I remember having some problem with BT (probably some my fault).
But when i try flash some blink code it works.

Unfortunately I can't help more than that.

ds2k5, 2023-11-23:
thanks....my colleage rent my his: SmartRF
so I could flash the hex file with the TI-Software on a Windows Box

JonRobert, 2023-12-01:
I've had good luck using a NodeMCU v2 using pins:
// define pins for NodeMCU
   int DD    = 14;  // data
   int DC    = 4;   // Clock
   int RESET = 5;
   int led   = 2

However now I'm trying to use a SEEED XIAO ESP32 C3 (because they are small and I have a couple of extra ones.
I'm using Pins:
// define pins for XIAO esp32-c3
   int RESET = 7;    //D5;    // i2c SCL
   int DC    = 6;    //D4;    // i2c SDA
   int DD    = 8;    //D8;    // SPI SCLK
   // int LED   = ;

The results seems to "hang" after (or trying to ) write the 1st block.
Any thoughts' 

```

Waiting for respond from Arduino...

Uploading firmware...

1

```